TheThingsNetwork/lorawan-stack #7547: Improve error flow when claiming a managed gateway

johanstokking, 2025-03-12:
### Summary

Improve the error flow when failing to claim a managed gateway.

### Current Situation

Currently, if a managed gateway cannot be claimed, the error message is conveyed from The Things Gateway Controller to the Console. The user is expected to read the error message, understand it and figure out what the course of action is to resolve it.

### Why do we need this? Who uses it, and when?

Users claiming managed gateways, e.g. The Things Indoor Gateway Pro, whenever that fails. This could be a typo in the EUI, a typo in the owner token or a owner token, no subscription attached and active for the gateway or the owner token is a gateway activation code and it has expired.

### Proposed Implementation

Check the error code from TTGC:

- `NotFound`: the gateway doesn't exist. This is most likely a typo in the EUI. An edge case is that The Things Stack is configured for the staging environment
- `FailedPrecondition`: the owner token is correct, but either there's no gateway subscription attached and active, or it is a correct gateway activation code and it has expired. If you need to distinguish between this, check what the error message contains:
    - `activation code expired`
    - `gateway subscription not attached and active`
    - add a sane default case if you don't match these error messages
- `PermissionDenied`: the owner token is invalid

Check also if/how the error is conveyed from DCS to the Console and if the error code is passed properly.

This should render a constructive message in the Console, potentially with documentation links (e.g. when the subscription is not attached and active, how to do that).

### Contributing

- [x] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [x] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

PavelJankoski, 2025-04-23:
> Check also if/how the error is conveyed from DCS to the Console and if the error code is passed properly.

@johanstokking It seems like the error is not conveyed correctly. It always returns `409 Conflict` as status code on errors and here are two different errors:

[image]

1.
```
{
    "code": 10,
    "message": "error:pkg/deviceclaimingserver:claim gateway (claim gateway)",
    "details": [
        {
            "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
            "namespace": "pkg/deviceclaimingserver",
            "name": "claim gateway",
            "message_format": "claim gateway",
            "correlation_id": "df2e759f537b41d7a882dd5b4c79e0d6",
            "cause": {
                "message_format": "not found: claim gateway",
                "correlation_id": "b87026440f9e4445bc26e03889dab1a1",
                "code": 5
            },
            "code": 10
        }
    ]
}
```

2.
```
{
    "code": 6,
    "message": "error:pkg/deviceclaimingserver:gateway_already_exists (gateway with EUI `EC656EFFFE000120` already exists)",
    "details": [
        {
            "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
            "namespace": "pkg/deviceclaimingserver",
            "name": "gateway_already_exists",
            "message_format": "gateway with EUI `{eui}` already exists",
            "attributes": {
                "eui": "EC656EFFFE000120"
            },
            "correlation_id": "20331145967a478bba7d470c0db2d156",
            "code": 6
        }
    ]
}
```

johanstokking, 2025-04-24:
> Check the error code from TTGC:
> 
>     * `NotFound`: the gateway doesn't exist. This is most likely a typo in the EUI. An edge case is that The Things Stack is configured for the staging environment

Try this with `EC656EFFFEFFFFFF`.

> * `FailedPrecondition`: the owner token is correct, but either there's no gateway subscription attached and active, or it is a correct gateway activation code and it has expired. If you need to [...]

Try this with `EC656EFFFE000120` and owner token `hbdsqkoa`. This should now work, I unclaimed the gateway.

Forget about `activation code expired`, we are changing the concept of gateway activation codes.
 
>     * `PermissionDenied`: the owner token is invalid

Try this with `EC656EFFFE000120` with any other owner token.

PavelJankoski, 2025-07-14:
This is tested and complete.