TheThingsNetwork/lorawan-stack #7566: Generate DevEUI button greyed out when registering more devices of the same type

StevenCellist, 2025-03-22:
### Summary

When registering a bunch of devices with TTN-provided DevEUIs (thank you! stupid device manufacturer doesn't provide a true DevEUI...), the button to generate DevEUIs is greyed out after the first device.

[image]
(keys set to 0 for the screenshot, otherwise would be a bit stupid ;) )

### Steps to Reproduce

1. Manually fill in credentials by entering JoinEUI, AppKey, (NwkKey) - generate the DevEUI using the button.
2. Set the radio button to "Register another device of the same type"
3. Register end device - the DevEUI field is cleared but the Generate button remains greyed out.

### Current Result

Cannot generate another DevEUI - must reset the JoinEUI for it to re-activate. Not the end of the world for sure, but breaks up the otherwise great flow of "Register another device".

### Expected Result

Make the Generate button available again after device is registered.

### Relevant Logs

```shell

```

### Deployment

The Things Stack Community Edition

### The Things Stack Version

3.34.0

### Client Name and Version

```shell

```

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2025-04-23:
This seems valid. @ryaplots: can you reproduce this?

ryaplots, 2025-04-28:
looking into this

ryaplots, 2025-04-28:
I reproduced. Will open a pr with a fix

ryaplots, 2025-05-08:
A fix has been merged but not yet deployed