TheThingsNetwork/lorawan-stack #7567: Missing "Name and ID" column for screen width of 576px or less (End device overview)

StevenCellist, 2025-03-22:
### Summary

For screen width equal to or smaller than 576px, the first column "Name and ID" column of the Application's 
End devices overview disappears without any way of seeing which device is which. (The problem already happens from e.g. 650 pixels and lower, depending on name/id length.) There is no horizontal scrolling to compensate for this.

### Steps to Reproduce

1. Check on mobile, or
2. Inspect on browser and resize screen width.

### Current Result

Name and ID are invisible (the column name overlaps with DevEUI even) - only the DevEUI/JoinEUI fields are visible. Interestingly, for widths <576px, horizontal scrolling does become available to see the Last seen column, but no way to see the Name and ID.

### Expected Result

Be able to see Name and ID column on smaller screens (mobile).

### Relevant Logs

```shell

```

### Deployment

The Things Stack Community Edition

### The Things Stack Version

v3.34.0

### Client Name and Version

```shell

```

### Proposed Fix

Switch to horizontal scrolling below certain screen width, keeping Name and ID column intact. The dilemma is to decide what the minimum width of the Name and ID column should be - the devices in my app only have an ID of length 8, but some people have much longer names... so not sure at which point horizontal scrolling should kick in vs decreasing width of Name and ID column.

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing a fix after the proposal above is approved.
- [ ] I can help by testing the fix before it's released.

### Validation

- [ ] The fix is tested in a staging environment.
- [ ] The fix is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2025-04-23:
Could you please add a screenshot? I just tested with a width of 428px and I can see the Name and ID column.

[image]

StevenCellist, 2025-04-24:
Looks like this is already tackled in a recent upgrade, I can't reproduce this anymore either and it looks all good. Thanks!

closed 2025-04-24 by StevenCellist