TheThingsNetwork/lorawan-stack #7594: MAC settings profile Console UI

halimi, 2025-04-15:
### Summary

We implemented the MAC settings profile API in https://github.com/TheThingsNetwork/lorawan-stack/issues/7380  for batch of devices it is useful but for a couple of devices it would be easier to use the Console UI.

### Current Situation

Currently we can manage MAC settings profiles via the API.

### Why do we need this? Who uses it, and when?

To have an easy way to manage MAC settings profiles and to associate/disassociate it to devices.

### Proposed Implementation

A MAC settings profile is bind to an application. You can create/update/list/delete MAC settings profiles. We need to create a new section under the application where you can manage the MAC settings profiles.

We need to make sure that you can assign a MAC settings profile to an individual end device. For this probably the best place to put it under the `end devices -> general settings -> network layer`. Currently you can find here the `Advanced MAC settings`. When you assign a MAC settings profile then we are removing the `Advanced MAC settings`, they are mutually exclusive so it should reflect on the UI as well. We need to either show the `MAC setting profile ID` or the `Advanced MAC settings`. When unassigning the MAC settings profile from the device (setting the profile ID to null) then we need to show the `Advanced MAC settings` again.

There is also a possibility to assign MAC settings profile to batch of end devices. Probably a good place  to put this under the `Application -> MAC settings profile` (the newly created page). Here you need to choose a `MAC settings profile` and you need to add a list of end-device IDs. You can also unassign the MAC setting profile from this list of end devices.

### Contributing

- [x] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [x] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2025-04-15:
@pierrephz: Please sync with @halimi for the views. Once done, we can implement the screens for `v3.35.0` maybe.

halimi, 2025-05-01:
> There is also a possibility to assign MAC settings profile to batch of end devices. Probably a good place to put this under the `Application -> MAC settings profile` (the newly created page). Here [...]

During the design session, it turned out that currently doesn't make sense to use this on the UI, because we don't have a possibility to get a list of end devices that are using the profile. So we cannot show here and makes it hard to change or modify the list of end devices that are using the profile.

We decided that you can assign/unassign a profile on a particular end device page only.

pierrephz, 2025-05-15:
## Here are mockups for the `MAC settings profile settings` (list of profiles + creation of profile)

Adding a new menu item in the Application sidebar where you can see existing profiles and create new ones:

[image]

When clicking on the button `Create MAC settings profile`, you go to the creation process:

[image]

**These mockup are a first version and open to feedback**

## Here are mockup of the MAC settings profile usage form the `End device settings` page:

[image]

[image]

[image]

[image]

[image]

[image]

These mockups are a first version and open to feedback cc @halimi @KrishnaIyer

ryaplots, 2025-08-19:
I have some questions about this. 
1. Considering this is for v3 I don't see why we need to redesign the form, we can rather implement this design in v4.
2. Our current modals do not allow for adding forms inside. To allow for this we would either have to refactor the whole modal system or create a new one specifically for this case. This is a lot of work, when we can simply redirect the user to the create form, just like we do with wifi profiles.
3. The advanced mac settings that we currently have in the frontend is a rather complex form, with over 30 fields, we show fields conditionally depending on ACTIVATION MODE, LoRaWAN version and the class of the device. If these profiles are created at the level of the application, then we do not have access to these values per device. So I assume that the mac settings profile form has to have these fields:
```
message MACSettings {
  option (thethings.flags.message) = {
    select: true,
    set: true
  };
  // Maximum delay for the device to answer a MAC request or a confirmed downlink frame.
  // If unset, the default value from Network Server configuration will be used.
  google.protobuf.Duration class_b_timeout = 1;
  // Periodicity of the class B ping slot.
  // If unset, the default value from Network Server configuration will be used.
  PingSlotPeriodValue ping_slot_periodicity = 2;
  // Data rate index of the class B ping slot.
  // If unset, the default value from Network Server configuration will be used.
  DataRateIndexValue ping_slot_data_rate_index = 3;
  // Frequency of the class B ping slot (Hz).
  // If unset, the default value from Network Server configuration will be used.
  ZeroableFrequencyValue ping_slot_frequency = 4;
  // Frequency of the class B beacon (Hz).
  // If unset, the default value from Network Server configuration will be used.
  ZeroableFrequencyValue beacon_frequency = 25;

  // Maximum delay for the device to answer a MAC request or a confirmed downlink frame.
  // If unset, the default value from Network Server configuration will be used.
  google.protobuf.Duration class_c_timeout = 5;

  // Class A Rx1 delay.
  // If unset, the default value from Network Server configuration or regional parameters specification will be used.
  RxDelayValue rx1_delay = 6;
  // Rx1 data rate offset.
  // If unset, the default value from Network Server configuration will be used.
  DataRateOffsetValue rx1_data_rate_offset = 7;
  // Data rate index for Rx2.
  // If unset, the default value from Network Server configuration or regional parameters specification will be used.
  DataRateIndexValue rx2_data_rate_index = 8;
  // Frequency for Rx2 (Hz).
  // If unset, the default value from Network Server configuration or regional parameters specification will be used.
  FrequencyValue rx2_frequency = 9;
  // List of factory-preset frequencies.
  // If unset, the default value from Network Server configuration or regional parameters specification will be used.
  repeated uint64 factory_preset_frequencies = 10 [(validate.rules).repeated.max_items = 96];

  // Maximum uplink duty cycle (of all channels).
  AggregatedDutyCycleValue max_duty_cycle = 11;

  // Whether the device supports 32-bit frame counters.
  // If unset, the default value from Network Server configuration will be used.
  BoolValue supports_32_bit_f_cnt = 12;

  // Whether the Network Server should use ADR for the device.
  // This field is deprecated, use adr_settings instead.
  BoolValue use_adr = 13 [deprecated = true];
  // The ADR margin (dB) tells the network server how much margin it should add in ADR requests.
  // A bigger margin is less efficient, but gives a better chance of successful reception.
  // This field is deprecated, use adr_settings.dynamic.margin instead.
  google.protobuf.FloatValue adr_margin = 14 [deprecated = true];
  // Whether the device resets the frame counters (not LoRaWAN compliant).
  // If unset, the default value from Network Server configuration will be used.
  BoolValue resets_f_cnt = 15;
  // The interval after which a DevStatusReq MACCommand shall be sent.
  // If unset, the default value from Network Server configuration will be used.
  google.protobuf.Duration status_time_periodicity = 16;
  // Number of uplink messages after which a DevStatusReq MACCommand shall be sent.
  // If unset, the default value from Network Server configuration will be used.
  google.protobuf.UInt32Value status_count_periodicity = 17;

[... 41 lines cut ...]
  BoolValue uplink_dwell_time = 32;
  // Whether downlink dwell time is set (400ms).
  // If unset, the default value from Network Server configuration or regional parameters specification will be used.
  BoolValue downlink_dwell_time = 33;

  // Adaptive Data Rate settings.
  // If unset, the default value from Network Server configuration or regional parameters specification will be used.
  ADRSettings adr = 34;

  // Whether or not downlink messages should be scheduled.
  // This option can be used in order to disable any downlink interaction with the end device. It will affect all types
  // of downlink messages: data and MAC downlinks, and join accepts.
  BoolValue schedule_downlinks = 35;

  // The relay settings the end device is using.
  // If unset, the default value from Network Server configuration or regional parameters specification will be used.
  RelaySettings relay = 36;

  // The relay settings the Network Server should configure device to use via MAC commands.
[... cut ...]  Server configuration or regional parameters specification will be used.
  RelaySettings desired_relay = 37;
}
```
And the user fills in the ones they need? Is there validation on the back to prevent some fields from being accepted under some conditions, for example, `beacon_frequency` is the frequency of the class B beacon, if the mac settings profile has a value for this field but the profile is being set to a class C device, will it be possible to set this profile anyway?
@KrishnaIyer @halimi

halimi, 2025-08-19:
@ryaplots The idea is that when you create a MAC settings profile at that moment is not attached to any device so it is a general profile and should be able to set all the settings. Because it is not attached to a device therefore there is no backend and frontend validation for the set of settings.

The validation happens when you assign it to a device because that is the point when we know the device activation mode, lorawan version, class, etc. The backend will validate the set of settings and reject it if not fits for the device. In this case the user needs to change the settings in the profile to able to apply it.

ryaplots, 2025-08-19:
I see, that's clear! Thanks!

ryaplots, 2025-10-27:
I tested on staging and seems to be working well.