TheThingsNetwork/lorawan-stack #7600: And CN470 Frequency Plan of RP002 Regional Parameters

shiwzhi, 2025-04-23:
### Summary

Be able to register CN470 end device with  RP002 Regional Parameters frequencies

### Current Situation

Currently there is only RP001 Regional Parameters for CN470 frequency plan available in the console

[image]

### Why do we need this? Who uses it, and when?

End device that uses RP002 regional parameter

### Contributing

- [ ] I can help by doing more research.
- [ ] I can help by implementing the feature after the proposal above is approved.
- [x] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2025-04-23:
@shiwzhi. Frequency plans are layered on top of Regional parameter settings. You can choose the regional parameters using the **Regional Parameters version** setting.

[image]

StevenCellist, 2025-04-24:
I am pretty sure OP means RP002-1.0.4 which has a massively different version of CN470.