TheThingsNetwork/lorawan-stack #7628: Allow importing more than 100 devices using the CLI tool

vlasebian, 2025-05-22:
### Summary

Importing more than 100 devices using the CLI tool is restricted by rate limiting. It would be useful for a user to be able to import a larger number of devices without splitting the import file into smaller ones and processing each file separately.

### Current Situation

When trying to import a large number of end devices from the CLI tool, rate limiting restricts this at 100 devices:

```
error:pkg/ratelimit:rate_limit_exceeded (rate limit of `100` accesses per minute exceeded for resource `grpc:method:/ttn.lorawan.v3.DeviceRepository/GetTemplate::token:E5MPAAJ5S3PHGJREBBPOPJRHLEPEUDKHR2CEIHI`)
    rate=100
```

### Why do we need this? Who uses it, and when?

This is feature would be used by users when importing a large number of end devices into The Things Stack. It simplifies the process of importing end devices.

### Proposed Implementation

Add a pause & retry for the CLI tool when importing a large number of devices.

### Contributing

- [x] I can help by doing more research.
- [x] I can help by implementing the feature after the proposal above is approved.
- [x] I can help by testing the feature before it's released.

### Validation

- [ ] The feature is tested in a staging environment.
- [ ] The feature is documented in [The Things Stack Documentation](https://www.thethingsindustries.com/docs/)

### Code of Conduct

- [x] I agree to follow TTN's [Community Code of Conduct](https://github.com/TheThingsNetwork/Manifesto/blob/master/CodeOfConduct.md).

KrishnaIyer, 2025-05-26:
> Importing more than 100 devices using the CLI tool is restricted by rate limiting

100 is a magic number set on TTS Cloud/Sandbox clusters and is subject to change. Indeed as you mentioned in the issue, this should not consider the number of requests but backoff based on the HTTP rate-limitd (429) response.

KrishnaIyer, 2025-05-26:
This is a good one for @happyRip with his operational experience.

PR #7644 (2025-06-05): Allow importing more than 100 devices using the CLI tool

#### Summary

Closes #7628

#### Changes

- Add retry logic to API
- Use retry logic in end device create calls

#### Testing

##### Steps

1. Create a `devices.json` file with 10 random devices:

<details><summary><code>devices.json</code></summary>

```json
{
  "ids": {
    "device_id": "eui-1ae653c29605251a",
    "dev_eui": "1AE653C29605251A",
    "join_eui": "30A71376812205CE"
  },
  "lorawan_version": "MAC_V1_0_3",
  "lorawan_phy_version": "PHY_V1_0_3_REV_A",
  "frequency_plan_id": "EU_863_870",
  "supports_join": true,
  "root_key [... cut]